tessel/t2-docs #14: API example library

Frijol, 2015-10-17:
It would be cool if we had a library of quick examples for how to use different APIs.
A good starting set: PubNub, Twilio, Plotly, Firebase, ...

I'm envisioning a setup whereby we have a JSON linking to hackster embed links

(example of hackster embedded: from http://tessel.github.io/t2-start/modules/ambient.html

[image]
)

And then a view that pulls from there to some grid layout on tessel.io/docs.

This makes it easy for people who want to showcase their APIs to quickly make an example and show it off on our site, which lends itself to lots of great examples and options for our community.

Related: https://github.com/tessel/t2-start/issues/4

tcr, 2015-10-17:
We had some requests this weekend (might be a bit more complex):

* iotivity
* Pouchbase

Frijol, 2016-09-02:
This issue requires both website layout design (how to display the examples?) and repo design (how do we make it easy for people to add new examples to this repo?)