tessel/t2-docs #20: Add Fritzing part for Accelerometer Module

Frijol, 2015-10-18:
Find schematics etc. at https://tessel.io/opensource

mittalshravika, 2017-07-28:
Do we need to make fritzing diagrams like the one [here](https://github.com/tessel/t2-docs/issues/31) for climate module?

Frijol, 2017-07-29:
Yes, that's exactly right!

@rwaldron do we already keep Fritzing source in a Tessel repo? If not, we should figure out a place where they go.

mittalshravika, 2017-07-29:
Which software was used for the climate module [here](https://github.com/tessel/t2-docs/issues/31)?

Frijol, 2017-07-29:
should be http://fritzing.org/download/

rwaldron, 2017-08-02:
I made the svg in https://graphic.com/mac/ and imported it to fritzing

mittalshravika, 2017-08-02:
Is there any alternative for Windows?

rwaldron, 2017-08-02:
https://inkscape.org/en/ ?

mittalshravika, 2017-08-03:
[image]

@Frijol, @rwaldron, I tried making one for the accelerometer. Does this look fine?

Frijol, 2017-08-04:
Yes, looks good to me!

Frijol, 2017-08-04:
Per http://fritzing.org/parts/ it looks like the resolution of this issue would be to submit the part to http://forum.fritzing.org/c/parts-submit

mittalshravika, 2017-08-04:
So, should I submit the part on the link provided?

Frijol, 2017-08-04:
I think so! I haven't done any Fritzing before, so I'm not 100% sure, but it's what I would do.