tessel/t2-docs #31: Add Fritzing part for Servo Module

Frijol, 2015-10-18:
Find schematics etc. at https://tessel.io/opensource

rwaldron, 2015-10-18:
Need to make sure that the component uses the same dimensions, like: 

[image]

Source for that is here: http://johnny-five.io/img/breadboard/temperature-SI7020.fzz

And... https://github.com/rwaldron/tessel-io/tree/master/fritzing

Frijol, 2015-10-18:
oh nice, didn't realize you'd done that! relevant https://github.com/tessel/t2-docs/issues/25

rwaldron, 2015-10-18:
When I get to Boston I will upload the svgs so the component layers can be reused in other components

mittalshravika, 2017-08-03:
[image]

@Frijol, @rwaldron, I tried making one for the accelerometer. Does this look fine?