tessel/t2-docs #45: Need documentation for pin interrupts capabilities

johnnyman727, 2016-01-25:
We don't currently list which pins can receive interrupts. Based on [this line](https://github.com/tessel/t2-firmware/blob/master/node/tessel-export.js#L241), it looks like pins 2, 5, 6, and 7 support interrupts. The support interrupts types [are listed here](https://github.com/tessel/t2-firmware/blob/master/node/tessel-export.js#L422-L428).

HipsterBrown, 2016-05-25:
For anyone who wants to claim this issue:

As @johnnyman727 mentioned, and verified by [this pin table](https://tessel.io/docs/hardwareAPI#pin-mapping), pins 2, 5, 6, and 7 on both ports are `interrupt` capable. Instead of manually polling connected sensors/buttons/switches for a change in state, setting a pin interrupt will emit an event on that pin based on the `mode` we listen for. You can read more about interrupts and why they're useful here -> https://www.sparkfun.com/tutorials/326

Interrupt types:

- `rise`: pin voltage rises
- `fall`: pin voltage falls
- `change`: pin voltage changes
- `high`: can only be used by `pin.once()`, pin voltage is high
- `low`: can only be used by `pin.once()`, pin voltage is low

An example use of using a pin interrupt with Tessel:

```javascript
// Import the interface to Tessel hardware
var tessel = require('tessel');

// the button is connected to pin 2 on port A
const button = tessel.port.A.pin[2];

// the green LED will be our status light
const green = tessel.led[2];

// when the pin voltage drops, this event will be triggered and the green LED will toggle on or off.
button.on('fall', () => {
  green.toggle();
});
```

Fritzing:
[image]

HipsterBrown, 2016-05-31:
Fixed by #69

closed 2016-05-31 by HipsterBrown