tessel/t2-docs #56: Add required ports

huseyinkozan, 2016-04-14:
After have some problem listing and using the t2 over LAN [0], I tried to disable the firewall at Ubuntu 14.04 and it solved my problem. 

Shouldn't we add required ports (mdns for ex.)  to the docs ?

[0] https://github.com/tessel/t2-cli/issues/650

johnnyman727, 2016-04-15:
@huseyinkozan great job investigating this. I think we should definitely add this information to the docs. Can you provide a little more detail about how you disabled the firewall?

mDNS uses port 5353, I believe. Is there a way to open up comms only on that port?

huseyinkozan, 2016-04-15:
To disable the firewall :
```bash
sudo ufw disable
```
To open a port (for ex:5353) using *ufw*:
```bash
sudo ufw allow 5353
```
This will do same thing:
```bash
sudo ufw allow mdns
```

Here the detailed Ubuntu ZeroConf HowTo page:
https://help.ubuntu.com/community/HowToZeroconf

And there is another port (3702) at ZeroConf wiki page that I want to try. I will give you info after try it:
https://en.wikipedia.org/wiki/Zero-configuration_networking

johnnyman727, 2016-04-15:
@HipsterBrown more info on mDNS debugging.

huseyinkozan, 2016-04-15:
Tried these ports but no luck:
5353, 5350, 5351, 1900, 5297, 5298, 515, 53, 139, 445, 68, ... and some others.

Investigating ...

https://support.apple.com/en-us/HT202944

Frijol, 2017-02-04:
Is this issue still in need of resolution?

huseyinkozan, 2017-02-04:
Dont know. I am not depend on it.