tessel/t2-docs #57: Arguments are illegible in near-white on white background

sdaitzman, 2016-04-17:
Extremely low contrast ratio on that `(error, value)` after `function`:

[image]

johnnyman727, 2016-04-18:
Yikes, that's no good.

I think @Frijol is in the midst of transitioning to gitbook so this problem be resolved pretty soon.

Frijol, 2016-04-18:
oof, yeah. I loathe our current docs style (I think it's my fault anyway). @sdaitzman any thoughts on the Gitbook idea or associated PR? https://github.com/tessel/t2-docs/issues/50

sdaitzman, 2016-04-20:
The new style is beautiful, especially on those tables! 😄

closed 2016-04-20 by sdaitzman