tessel/t2-docs #63: Replace master/slave references with primary/replica

HipsterBrown, 2016-04-28:
There are many uses of the master/slave terminology within the [hardware API](https://tessel.io/docs/hardwareAPI) and [communication protocols](https://tessel.io/docs/communicationProtocols) documentation. Much like the [django project](https://github.com/django/django/pull/2692) and [many others](https://en.wikipedia.org/wiki/Master/slave_%28technology%29#Appropriateness_of_usage), we should replace these references with more appropriate and accurate wording, i.e. primary/replica.

LinusU, 2016-04-28:
Shouldn't we pick something else than primary/replica though? That doesn't seem accurate at all, it only works for databases where there is actual replication of data going on. In this case, the master is the one in charge of the communication...

HipsterBrown, 2016-04-28:
@linusu I'm open to suggestions. I've heard of primary/secondary as well.

reconbot, 2016-04-28:
Host / client seems to fit the model. Or controller/ client

rwaldron, 2016-04-28:
"primary" and "replica" do not accurately describe the roles that you wish to use them as replacements for. The changes made in Django _do_ make sense and **more accurately** describe the roles they represent, ie. database replication (see, the word is right there). Oxford Dictionary provides clear definitions of the terms "master" and "slave" <sup>[1]</sup> that are contextually appropriate: 

master: 

> A machine or device directly controlling another

slave: 

> A device, or part of one, directly controlled by another

It would be inappropriate and inaccurate to call these "primary/replica" or "primary/secondary" or "host/client", and all uses of "master" and "slave" in the docs are contextually appropriate and accurate. Specifically, I2C and/or SPI slave devices are not a "replica", "secondary" or "client" of anything, they are, _by definition_ "A device, or part of one, directly controlled by another". Changing these terms will also create an unnecessary translation barrier for anyone that's ever read anything at all about these protocols. Does changing these from "master" to "slave" also mean changing the protocols themselves? Does MISO become PIRO or HICO or CICO, MOSI to RIPO or COHI or COCI? I don't think so. There is no known definition that supports Django's previous use of "master" and "slave", inarguably that use was a catachresis—just as it would be to substitute words which are in a protocol, but without applying the same substitutions to the protocol. Concretely: if you wish for someone with a background in wire communication protocols to understand these docs, you shouldn't just make up new names for things. 

[0]: http://www.oxforddictionaries.com/us/definition/american_english/master#master__7
[1]: http://www.oxforddictionaries.com/us/definition/american_english/slave#slave__7

HipsterBrown, 2016-04-28:
@rwaldron Given those definitions, I agree that primary/replica or host/client are not accurate replacements. However, I am sure we can find more inclusive and accessible replacements, like controller / client or leader / follower. 

> Does changing these from "master" to "slave" also mean changing the protocols themselves? Does MISO become PIRO or HICO or CICO, MOSI to RIPO or COHI or COCI?

No, that's not what I'm suggesting. We can link to the original standards that use this terminology, as well as consider using [the alternative naming conventions](https://en.wikipedia.org/wiki/Serial_Peripheral_Interface_Bus#Interface) for those interfaces, i.e. MISO === DOUT.

> if you wish for someone with a background in wire communication protocols to understand these docs, you shouldn't just make up new names for things.

Since this is a relevant concern, we can reference our use of these aliased terms, either by linking to this issue or a PR with those changes with context behind why those changes would be made. 

As part of making the Tessel project an inclusive community, we can make considerations for folks where master / slave carry more meaning than the definitions provided by Oxford Dictionary.

rwaldron, 2016-04-28:
> Since this is a relevant concern, we can reference our use of these aliased terms, either by linking to this issue or a PR with those changes with context behind why those changes would be made.

This means the words "master" and "slave" will still exist in the docs. 

> for folks where master / slave carry more meaning than the definitions provided by Oxford Dictionary.

I recommend simply including the Oxford Dictionary definition, as well as the links I've provided, before the words appear in content. 

[image]

PR #65 (2016-05-02): Adds 'Terms Used' section 
Fixes #63 

Acknowledges that `master`/`slave` are not terms chosen by the Tessel project, instead they are used as references to the hardware industry standard.
closed 2016-05-02 by HipsterBrown