tessel/t2-docs #67: pin.analogWrite(0.6) sets voltage to 0.6V, not 60% of 3.3V

smrq, 2016-05-17:
According to [the docs](https://github.com/tessel/t2-docs/blob/8c94789d1eaa718921f8e32dca47e447056491a3/hardware-api.md#analog-pins):

> pin.analogWrite(0.6);  // turn pin to 60% of high

But I found that rather than ~2V, it set the pin to 0.6V.  Not sure if it's a docs problem or a software problem -- but I did find it surprising (I expected `analogWrite(1)` to set the pin to 3.3V, not 1V.)

johnnyman727, 2016-05-20:
I agree that this is confusing. I ran into this recently as well.

I recommend that we update the firmware to use `0.0-1.0` for consistency.

Frijol, 2016-06-05:
Discussing also in https://github.com/tessel/t2-firmware/issues/175

Frijol, 2016-09-02:
resolved with https://github.com/tessel/t2-firmware/pull/184

closed 2016-09-02 by Frijol