tessel/t2-docs #80: SPI example has non-sensical inclusion of math expression that's actually longer than the number it represents

rwaldron, 2016-07-15:
```js
var spi = new port.SPI({
  clockSpeed: 4*1000*1000, // 4MHz
  cpol: 1, // polarity
  cpha: 0, // clock phase
});
```

Is there a reason for assigning `clockSpeed` the value of an expression to be evaluated, versus just the number?

```js
var spi = new port.SPI({
  clockSpeed: 4000000, // 4MHz
  cpol: 1, // polarity
  cpha: 0, // clock phase
});
```

Also, we should deprecate (with aliasing) `clockSpeed`, `cpol` and `cphase` in favor of parameters that are named something less hostile.

Recommending: 

| Old Name | New Name |
| ---------------|----------------|
| clockSpeed | frequency |
| cpol | polarity |
| cpha | phase |

jandrieu, 2016-09-07:
Representing large numbers like 4,000,000 as ```4*1000*1000``` instead of ```4000000``` makes them easier to read, debug, and maintain.

I'd keep the clockspeed as it is, but I agree more readable property names for polarity and phase would be an improvement.

tcr, 2016-09-07:
If we were using Java/Python/Rust, representing it as 4_000_000 would be ideal. ;) Since this is for docs, it probably if we show people clock speed is just a number. Persons maintaining their own code can use 4*1000*1000.

+1 to polarity and phase. Where did terms CPHA and CPOL even come from? If it's restricted to Arduino examples then it makes sense to deprecate them.

I feel like "clockSpeed" is more applicable than "frequency" but YMMV.

Smadattam, 2017-12-09:
Hi guys, I'm new here, and am a pretty green programmer. Would it make sense to create a constant titled MHz (or some other descriptive name) instead, and set it equal to 1000*1000? That way the expression could become "4*MHz" which could be even more readable.

arushi019, 2018-08-23:
Hi @frijol ! Can the issue be closed now that the pull request has been merged?

Frijol, 2018-09-10:
Yes, thanks for noticing!

closed 2018-09-10 by Frijol