tessel/t2-docs #101: Add an overview of Tessel testing infrastucture

Frijol, 2016-09-01:
_From @johnnyman727 on October 18, 2015 14:33_

_Copied from original issue: tessel/onboarding#10_

Frijol, 2016-09-01:
@johnnyman727 can you say more about this? Is this something that can/should belong in github.com/tessel/t2-docs?

johnnyman727, 2016-09-13:
I'm assuming that this is about software testing infrastructure and not hardware testing.

We've pretty much standardized on using `grunt` and `nodeunit` for our software testing infrastructure. I'm not sure it should be in `t2-docs` unless we specifically add a contributing section. If we do, we can recommend developers run `t2 crash-reporter off` to save our inboxes.

Frijol, 2016-09-14:
so far we have documented contributor-specific items in the repos. Maybe this should be mentioned in Contributing.md or on the CLI readme?

johnnyman727, 2016-09-14:
Yeah, I think leaving testing notes up to the individual repo makes sense, especially now that we have multiple languages going on. The [CLI already has ample testing docs](https://github.com/tessel/t2-cli#testing) so I think we can go ahead and close this issue.

closed 2016-09-14 by johnnyman727