tessel/t2-docs #97: T2 CLI docs do not include mention of `t2 root`

Frijol, 2016-09-01:


Frijol, 2016-09-01:
Btw, is there a better way to document CLI commands than here in this doc? Feels redundant with the documentation that's inherent to the CLI, e.g. you can just `t2 root -v` to learn in an up-to-date manner what is going on. cc @HipsterBrown maybe you have a thought about this

Frijol, 2016-09-02:
Discussed live, we decided it's still useful to manually update the docs since this provides a different sort of insight than that provided by the CLI documentation – "why does this exist, and what is it" rather than just "how do I do this"

Frijol, 2016-09-02:
The root command can be found in the CLI code (with some in-line docs) here: https://github.com/tessel/t2-cli/blob/master/lib/controller.js#L803

Jeff-Duke, 2016-10-28:
PR open - https://github.com/tessel/t2-docs/pull/130
@Frijol @HipsterBrown

commit 2071109a (2016-10-29): Merge pull request #130 from Jeff-Duke/t2-root

Document t2 root command in CLI docs - closes #97
closed 2016-10-29 by HipsterBrown