tessel/t2-docs #110: Document power management

dbuentello, 2016-09-04:
There is sparse documentation dealing with power for both contributors & users. [This](https://forums.tessel.io/t/power-management-of-tessel-2/1630) was a good start. 

Questions a user might have:
Whats the power consumption of the tessel2? ??
Is there a low power mode?

Questions a contributor might have: 
How is power distributed?

dbuentello, 2016-09-04:
Something like this would be end goal
Users: https://developer.mbed.org/cookbook/Power-Management
Contributor: https://developer.mbed.org/users/no2chem/notebook/mbed-power-controlconsumption/

Frijol, 2016-09-06:
duplicate of https://github.com/tessel/t2-docs/issues/4, @dbuentello want to add your comments to that issue & then I'll close this one?

dbuentello, 2016-09-06:
oops, guess I miss that. transferred notes. close at will.

Frijol, 2016-09-06:
👍

closed 2016-09-06 by Frijol