tessel/t2-docs #115: Would be nice to have hardware docs

johnnyman727, 2016-09-13:
Many people have been asking for more info on hardware-related specifications like maximum input voltage. It would be great to have T2 documentation [akin to existing T1 docs](https://github.com/tessel/hardware).

tcr, 2016-09-14:
In particular, I would love to see these moved to docs https://github.com/tessel/hardware/blob/master/modules-overview.md

johnnyman727, 2016-09-15:
We've had other people ask about GPIO output (current) and should include it in the docs.

[image]

Frijol, 2016-09-15:
Lots of good info here https://tessel.io/blog/113259439202/tessel-2-hardware-overview

closed 2016-09-22 by Frijol