tessel/t2-docs #126: Add Rust documentation

tcr, 2016-10-17:
We should update docs to include instructions from [tessel-rust](http://github.com/tessel/tessel-rust), now that we have an out of box experience for remote and local compilation with t2-cli.

richa-coder, 2016-10-18:
Can i start with this?

tcr, 2016-10-18:
@richa-coder Yeah absolutely!

We want to fill out the "Rust" section of the [Supported Languages](https://github.com/tessel/t2-docs/blob/master/API/Supported_Languages.md) page. The existing content is old and talks about what we planned to do for Rust support. We can scrap that.

Now that we added Rust support to Tessel's CLI, we want to give instructions on how to use it. This "Quickstart" section has all the information: https://github.com/tessel/tessel-rust

The Quickstart is good, but it would be even better if the documentation described what each command did:

* `cargo tessel sdk install` installs the Rust cross-compiler
* `t2 init --lang=rust` writes out the "blinking lights" example
* `t2 run blinky` runs the "blinky" binary on Tessel
* `t2 run blinky --rustcc` runs the "blinky" binary, compiling it using the cross-compilation server

If there's anything you want more detail on, just ask. The goal is to get the info in the right place and make it understandable to someone reading the guide for the first time.

richa-coder, 2016-10-18:
so you want me to remove rust section from Supported languages and just
give a bit explanation on each commands for the quickstart page?

[image: --]

Richa Singh
[image: https://]about.me/richa-singh
<https://about.me/richa-singh?promo=email_sig&utm_source=email_sig&utm_medium=email_sig&utm_campaign=external_links>

Frijol, 2016-10-21:
@richa-coder until we reach official full Rust support (we're not there yet) I think it's best if we keep the t2-docs information about Rust commands in the Supported Languages page and maintain the Work In Progess marker in the heading.

I think @tcr is suggesting that we pull information from the tessel-rust repo (specifically the Quickstart section) into the Supported Languages page so people don't have to search for it as much. And separately, it would be good to add documentation on the specific commands to the Quickstart in both locations. @tcr am I reading that right?