tessel/t2-docs #138: GPIO function diagram

perjg, 2016-11-29:
It would be nice to have an mbed style GPIO function diagram for the Tessel 2 board.

Here are two examples:
https://developer.mbed.org/media/uploads/GregC/xfrdm-kl25z_headers2.png.pagespeed.ic.altGnczqQ9.webp

https://developer.mbed.org/media/uploads/star297/xteens3.1-main.jpg.pagespeed.ic.XmOMXz7Ymr.webp

The diagram should show a picture of the Tessel 2 board, with each GPIO pin marked with pin name as well as other functions available on that pin (such as SDA0, SCL0 etc.)

Frijol, 2017-02-05:
Do you have a suggestion for software that can be used to build this?

perjg, 2017-02-12:
Inkscape is probably the best free software to use for this.

mittalshravika, 2017-07-09:
Hi, I tried solving this issue by making the pinout diagram for tessel.
[image]

PR #162 (2017-07-28): Pinout diagram
Fixes #138.
closed 2017-07-28 by Frijol