tessel/t2-docs #146: How do I know if I need to update section should mention auto-notifications

Frijol, 2017-01-26:
https://tessel.gitbooks.io/t2-docs/content/API/CLI.html#how-do-i-know-if-i-need-to-update-my-t2

The "How do I know..." section linked above should mention what you will get automatic updates about. Specifically, it should mention that:

* The CLI will tell you if you are running an old version of CLI if you are connected to the internet and have a version newer than (when did this get introduced?) when you run any t2 command
* The CLI will not tell you if you are running an old version of firmware

mittalshravika, 2017-07-24:
Hey, I worked on this Issue. @Frijol you may review the same.

mittalshravika, 2017-07-26:
Hey, I had a query that for this issue we need to mention that there is no automatic way in which the user can be made aware that there is a newer version for t2-firmware in the t2-docs?

Is anything else also required for resolving this?

Frijol, 2017-07-26:
I'm not sure I understand your question.

I would suggest:
* Mention how you'd know if t2-cli needs an update via the notifier
* Mention that you would not know if firmware or CLI needs an update but that the fastest way to find out is to run `t2 update`

mittalshravika, 2017-07-27:
Hey, I tried connecting wifi to the other Tessel we have, but it did not give an update notification for t2-cli as it had given earlier.

rwaldron, 2017-07-27:
> Hey, I tried connecting wifi to the other Tessel we have, but it did not give an update notification for t2-cli as it had given earlier.
>
> Is it because of that 24 hour functionality?

Are you referring to this: https://www.npmjs.com/package/update-notifier#updatecheckinterval ? If yes, then I think you're right.

Frijol, 2017-07-29:
reopened; incorrectly auto-closed with a PR merge

mittalshravika, 2017-07-31:
@Frijol, created a PR for this issue. You may review the same.

commit 21fa8e0e (2017-07-28): Removed the commits made for fix #146
PR #166 (2017-08-02): Fix for Issue - 146
Fixes #146. Updated the "How do I know if I need to update" section in t2-docs with update instructions for T2-CLI and T2-Firmware.
closed 2017-08-02 by Frijol