tessel/t2-docs #153: `t2 restore` is undocumented

Frijol, 2017-03-14:
`t2 restore` isn't documented in docs. CLI documents it as: "Restore your Tessel by installing the factory version of OpenWrt."

This is a very useful command! But I don't remember how it works, do you just plug in a tessel and run it without waiting for bootup? Do you have to put T2 into DFU mode?

Frijol, 2017-08-02:
DFU mode: "device firmware upgrade mode". On Tessel 2, I believe this is activated by holding down the button on the board while plugging in Tessel (normally, with USB). Lights will flash a bit differently, and you should be able to run T2 CLI commands (but not the usual ones)

rwaldron, 2017-08-02:
When running `t2 restore`, the command will put the Tessel 2 into DFU mode automatically—it _must_ begin in normal operation mode.

Jeff-Duke, 2017-10-04:
I'd like to grab this if it hasn't been picked up already.  I can run the command on my Tessel this weekend to observe how it goes and document the results.

HipsterBrown, 2017-10-04:
Thanks for picking this up @Jeff-Duke. You can ping me here or in the [Tessel Slack](https://slackin.tessel.io) if you have any questions.

Jeff-Duke, 2017-10-04:
Thanks @HipsterBrown!

Jeff-Duke, 2017-10-07:
I've opened PR #184 for this.

commit 2c84fe78 (2017-10-17): Merge pull request #184 from Jeff-Duke/t2-restore

Update CLI docs -- closes #153
closed 2017-10-17 by HipsterBrown