tessel/t2-docs #177: Improve 5V sourcing section

Frijol, 2017-08-15:
There is a very small mention that you can get 5V from T2 here: https://github.com/tessel/t2-docs/blob/master/Hardware/Tessel_2_Overview.md but it could stand to be more helpful, as here: https://forums.tessel.io/t/accessing-5v-on-tessel-2-powering-tessel-2-with-wires/2368

It may also be useful to note that you can source an external 5V using the Servo module's breakout (wise for applications that might require more current than T2 should safely provide, such as moving motors).

soapdog, 2017-08-15:
Another important highlight would be how to power Tessel 2 from 5vin, I know it is quickly described there but a linked tutorial would go a long way with people who are new to DIY electronics (something like iFixIt guides that contain copious amount of annotated high quality photos). 

I know this is more work but you can't never have too much documentation or photos 😸

mbrevda, 2018-06-26:
That is down. Is there anywhere else that information might be available?

mbrevda, 2018-07-08:
This is awesome and perfectly helpful, thanks!

arushi019, 2018-08-22:
HI @Frijol @HipsterBrown  ! What should we do to close this issue? We are not able to really understand here what to do.

Frijol, 2018-08-22:
Good question @arushi019! Here's what it takes to close this issue:

## 1. Make a new file in t2-docs to hold this information.

Should be under "Hardware" such that it shows up here in the sidebar of tessel.io/docs, even with Modules and Tessel 2 Overview:

[image]

– so, a new markdown file in this folder: https://github.com/tessel/t2-docs/tree/master/Hardware

It should be called something clear, short, and descriptive like "5V Power with Tessel".

(It could arguably go in the Tutorials folder instead. Which is more appropriate? You decide.)

## 2. Read and understand the notes above

Ideally, use the information above to:

* Source 5V from Tessel's main board
* Source 5V from a Servo module
* Power Tessel from a 5V source like a battery pack

Take pictures of the setup and/or make Fritzing diagrams.

However, that might be pretty time consuming. You don't have to do pictures and diagrams as long as you (a) understand everything and (b) are able to describe it clearly in writing. I'd happily merge a PR that just uses the diagram I made. Perfect is the enemy of good. Better to have something in the docs than nothing (and a really nice pull request in progress). Remember that you can always add improvements (pictures, more descriptions) later in a separate pull request if you find you have time.

## 3. Explain how to use 5V with Tessel in your markdown file

These three things must be clear: how to...

* Source 5V from Tessel's main board
* Source 5V from a Servo module
* Power Tessel from a 5V source like a battery pack

Minimally, you could copy/paste my comment and check it for spelling and clarity. But if you want to go further, here are improvements you could make:

* Give it a more formal, textbook-like tone
* Think about what people are looking for when they arrive at this page, make sure headings etc. direct them to what they need
* Give examples

...but again, start with the minimum. Make a PR that has just the basic stuff, and submit it. You can add more later, if there's time.

Hope this helps, and please ask more questions if they arise!