tessel/t2-docs #178: Document `pin.toggle()`

xasc, 2017-08-22:
[`pin.toggle()`](https://github.com/tessel/t2-firmware/blob/b7b19335032319be80beb1942b91b08fce8d158e/node/tessel-export.js#L831) sounds like a useful method for changing the value of digital GPIOs with defined level. There is currently no reference in the [Hardware API](https://tessel.gitbooks.io/t2-docs/content/API/Hardware_API.html).

Should this method be added to the documentation at all?

If so, I suggest to add method description and a short example to [Digital Pins](https://tessel.gitbooks.io/t2-docs/content/API/Hardware_API.html#digital-pins).

HipsterBrown, 2017-08-24:
I think this should definitely be documented because I agree it looks like a convenient and useful method. cc: @rwaldron @Frijol 

@xasc Do you feel comfortable opening a PR to document this API?

xasc, 2017-08-24:
Yes, I can take care of it.

commit ae3f7698 (2017-09-05): Merge pull request #180 from xasc/issue-178

Fixes #178 - pin.toggle() docs
closed 2017-09-05 by HipsterBrown