tessel/t2-docs #179: I2C Tutorial

HipsterBrown, 2017-08-31:
The I2C API methods are currently documented, but it would be nice to add a complete tutorial of how to use the API in a project. (API Reference -> https://tessel.gitbooks.io/t2-docs/content/API/Hardware_API.html#i2c). 

The documentation would go in the "Tutorials" directory in a file called `I2C.md`.

I think it would be nice to match the structure of the [PWM tutorial](https://tessel.gitbooks.io/t2-docs/content/Tutorials/Pulse_Width_Modulation.html) and [Pin Interrupt tutorial](https://tessel.gitbooks.io/t2-docs/content/Tutorials/Interrupts.html]). I think the tutorial circuit could use the [Tessel accelerometer module](https://github.com/tessel/accel-mma84) because we have working a library to reference.

Steps to complete:
- [ ] create a code sample that uses the I2C API to communicate write to and/or read from the accelerometer module
- [ ] add a working [Fritzing](http://fritzing.org/home/) example to match the code sample 
- [ ] add documentation of how the API communication works, i.e. why certain hex address are sent over and references to the datasheet for the component

Feel free to mention me in this issue, or a PR, or post in [Tessel Slack #community channel](https://tessel-slack.herokuapp.com/) with question or if review is needed.

mittalshravika, 2017-08-31:
Hey, @HipsterBrown, I was exploring this issue. So I tried reading from the accelerometer module using the reference code given [here](https://tessel.gitbooks.io/t2-docs/content/API/Hardware_API.html#i2c) and the I2C address for accelerometer module given [here](https://github.com/tessel/accel-mma84/blob/master/index.js#L15). 
After running the code I got the following output:
[image]
Is this what is required to be done for the code portion of this issue or something else has to be worked out?

tcr, 2017-08-31:
@mittalshravika The code for the I2C module is very messy, I apologize :(

Last time I wrote this example, I had something similar to your code. I believe the simplest demo is to write a byte for the WHO_AM_I register, 0x0D, then read one byte. So maybe something like this:

```
var tessel = require('tessel');

// Connect to device
var port = tessel.port.A; // Use the SCL/SDA pins of Port A
var slaveAddress = 0xDE; // Specific to device
var i2c = new port.I2C(slaveAddress); // Initialize I2C communication

// Details of I2C transfer
var numBytesToRead = 1; // Read back this number of bytes

// Send/recieve data over I2C using i2c.transfer
i2c.transfer(new Buffer([0x0D]), numBytesToRead, function (error, dataReceived) {
  // Print data received (buffer of hex values)
  console.log('Buffer returned by I2C slave device ('+slaveAddress.toString(16)+'):', dataReceived);
});
```

The returned buffer from the I2C slave device should be `[0x2A]`.

If `i2c.transfer` fails to work, try running `i2c.write(new Buffer([0x0D), function (err, value) { ... })` and then inside that callback, `i2c.read` 1 byte, and see what the buffer contains. This should again be 0x2A.

If all that works, you might have enough to get the example to read acceleration values. Ignore the rest of the setup methods and skip down to the `getAcceleration` function and see how it writes a register key and then reads six packed bytes of data.

mittalshravika, 2017-08-31:
Thanks a lot for the inputs!!
For accelerometer the slave address would be 0x1D? Just want to make sure that I have interpreted it correctly.

tcr, 2017-08-31:
Yes you have! I wrote it incorrectly in my post, but you had it correct in yours.

I believe it can also be 0x1C depending on if the two pads on the back of your module are connected. They almost certainly won't be though.

mittalshravika, 2017-09-02:
```js
var tessel = require('tessel'); //Import tessel

// Connect to device
var port = tessel.port.A; // Select Port A of Tessel
var slaveAddress = 0x1D; // Specefic for accelerometer module
var i2c = new port.I2C(slaveAddress); // Initialize I2C communication

// Details of I2C transfer
var numBytesToRead = 1; // Read back this number of bytes

i2c.read(numBytesToRead, function (error, dataReceived) {
  // Print data received (buffer of hex values)
  console.log('Buffer returned by I2C slave device ('+slaveAddress.toString(16)+'):', dataReceived);
});

// Read/Receive data over I2C using i2c.transfer
i2c.transfer(new Buffer([0x0D]), numBytesToRead, function (error, dataReceived) {
    // Print data received (buffer of hex values)
  console.log('Buffer returned by I2C slave device ('+slaveAddress.toString(16)+'):', dataReceived);
});

//Now, try to print the accelerometer data using i2c.transfer
i2c.transfer(new Buffer([0x01]), 6, function (error, dataReceived) {
    // Print data received (buffer of hex values)
    if (error) throw error;

    //Create a blank array for the output
    var out=[];
    for (var i=0;i<3;i++){ //iterating for the x, y, z values

      var gCount=(dataReceived[i*2] << 8) | dataReceived[(i*2)+1]; //Converting the 8 bit data into a 12 bit
      gCount=gCount >> 4;

      if (dataReceived[i*2] > 0x7F) {
          gCount = -(1 + 0xFFF - gCount); // Transform into negative 2's complement
        }

        out[i] = gCount / ((1<<12)/(2*2));
    }

    console.log('The x, y, z values are :',out); //Log the Array containing the x,y,z values
});
```
Is this piece of code fine for the tutorial?
Below is the output it gives:
[image]

brihijoshi, 2017-09-03:
@tcr @HipsterBrown

Frijol, 2017-09-03:
It looks like you're ready to start a PR for this issue! Good work so far on this.

mittalshravika, 2017-09-03:
Thanks a lot!! We would be sending a PR soon.