tessel/t2-docs #182: Pinpull Tutorial link broken

brihijoshi, 2017-09-17:
The tutorial that is made for pinpull is a markdown file, like Interrupts. However, the link that is provided in the Hardware API is an HTML file, like interrupts. While trying to click on the tutorial for pinpull, the link is broken. Is there any other way to build the gitbooks, so that it automatically renders the markdown files to HTML files?

@mittalshravika

Frijol, 2017-09-17:
I'm not sure I understand the issue. When the change is submitted to master, the _book file (which contains the html versions) is automatically generated by the Gitbook site. Can you perhaps give screenshots or line links for which URL is broken?

It does look like the tutorial link is missing from the sidebar, though.

Frijol, 2017-09-17:
Re the link missing from the sidebar, it looks like the new SUMMARY.md wasn't committed in https://github.com/tessel/t2-docs/pull/167 ?

brihijoshi, 2017-09-18:
[image]

This is what happens when I click on the link for the pinpull tutorial

brihijoshi, 2018-01-25:
Hi, this seems to be resolved now! Closing the issue

closed 2018-01-25 by brihijoshi